wpilibsuite/SystemcoreTesting #15: Debug Launch for Java not seeming to work

gerth2, 2025-06-24:
I'm able to click the button and I see happy looking messages, but the debugger fails to connect.

Here's a video:

https://photos.app.goo.gl/gYoPByeu9kiET7ax5

I'm happy to debug more, just not sure if this is known.

My Debug Console reports:

```
[Warn] The debugger and the debuggee are running in different versions of JVMs. You could see wrong source mapping results.
Debugger JVM version: 17.0.12
Debuggee JVM version: 21
```

ThadHouse, 2025-06-24:
Debug for systemcore targets has not been added to gradlerio yet. So I'm not sure how its connecting at all. This hasn't been tested at all.

gerth2, 2025-06-24:
All good! I'll skip that step for the moment. FWIW we usually made our own support in python, i'm expecting to do the same again here. Which is perfectly fine

Kevin-OConnor, 2025-06-25:
Apparently not as simple of a perms issue as I expected. Even changing the user of the systemd service to systemcore it still works when run manually and not when run by the service. Looking at the output via journalctl it looks like fmtlib is throwing an assert on a negative value: https://github.com/fmtlib/fmt/blob/0041a40c1350ba702d475b9c4ad62da77caea164/include/fmt/base.h#L433. Which feels like it may be an "error throwing an error" type situation.