wpilibsuite/SystemcoreTesting #17: Blank Display, Hardware Fault With USB

j0n5m1th, 2025-06-24:
The display can go blank when using a USB device like a CANivore. All status LEDs are solid on (specifically status, solid red), implying some hardware fault. The display will stay off and the status LED will show solid red after a power cycle.

Hardware configuration:
- SystemCore is powered the same way we would wire a roborio (follows R615.B)
- CANivore plugged into USB 0, no other USB devices
- CANivore wired for redundant power as described in the user guide
- "can_s0" effectively used as the "rio" bus which contains a mix of devices from various vendors
- Ethernet plugged directly into the "rio" port on the VH-109 (POE wires removed, follows R703.B)

System configuration:
- installed CANivore native packages declared in the CTR-Phoenix doc
- all CTRE devices on the available 2027 firmware
- all other devices on the system CAN bus are the latest 2025 versions

Software:
- project generated from 2027 WPILib VSCode and another 2025 ported project
- CTRE devices used on both the CANivore and s0 busses
- all non-CTRE devices are not referenced in code

We haven't been able to trace the issue yet (still fresh), but it does seem to be reproducible.

It's also worth noting that the robot did behave appropriately when enabled, so devices were still functional even though the display was not.

j0n5m1th, 2025-06-24:
I'm pretty sure a power cycle does not re-enable the display.

At one point, we reflashed SystemCore and the screen became functional again.

I've got a student that should be able to dig into this a bit more this afternoon.

j0n5m1th, 2025-06-25:
tldr; We tried to reproduce inadvertently disabling the display, but failed to do so. Re-flashing SystemCore was sufficient to re-enable the display and we could not disable the display or fault status (despite experiencing the issue yesterday).

Steps from today's efforts to reproduce the issue.

Screen was disabled (state from yesterday)
- Power cycle
- DFU bootup
- Power cycle
- Deploy blank project
- Power cycle
- Deploy working project
- Power cycle
- Run gradle script to kill robot process
- Power cycle
- Uninstall CANivore driver
- Power cycle
- Unplug CANivore
- Power cycle
- Remove ALL other wires including ethernet
- Power cycle
- Reflash unit with ethernet connected (no USB)
- Power cycle
- Reflash unit, (with only power wires connected)
- Many power cycles, the display is now re-enabled (each previous step the display was disabled), (one power cycle the screen displayed mangled text)

With a functional display:
- plugged back in all devices (ex. Ethernet, CANivore, can_s0)
- power cycle
- install CANivore native packages
- power cycle
- set team number
- power cycle
- push blank code project
- power cycle
- reference one CTRE device on CANivore
- power cycle
- push working robot code (from yesterday when the display was disabled)
- power cycle

At all times the display was functional. We went through this same general processultiple times. The screen was enabled at all times.

Grasping at straws, we tried to see if the CTRE vendordep from yesterday was a contributing factor (systemcore instead of linuxsystemcore platform). After a Gradle clean and pulling the "day old" vendordep, the display still functioned. We then tried to do this on a freshly flashed SystemCore to get a clean slate and the display was still enabled.

kevinwang, 2025-06-26:
A similar thing just happened to our unit at about 2.5 hours of uptime. The display started glitching out and after a power cycle, blank display and 3 solid LEDs.

https://github.com/user-attachments/assets/1144fb7c-b0a2-4682-aac5-9ab3d3a59e4e

[image]

j0n5m1th, 2025-07-02:
@bhjelstrom 

We lost the display again after a cold boot last weekend. We haven't been able to get it to re-enable. No measure of flashing, power cycling, code deploys, etc. seem to have an effect.

The good news is that the latest OS image does show the expected blink pattern:
[image]

Like Jonah, do you want this one back to investigate? We're hoping to run SystemCore at IRI next week & share with teams.

EthyMoney, 2025-07-02:
> Alpha 3 addresses this issue.
> 
> We were driving the display with about 2x the current we originally thought. This was still barely within spec, but it completely eliminated our safety margin.
> 
> Alpha 3 addresses this in a few ways:
> 
>     * Reduce display current by 80%. The screen will not be as bright. Let us know if the new brightness level is a problem
> 
>     * The display will automatically dim and shut down after 10 seconds of no-motion.
> 
>     * The display will turn on again if you tap your Systemcore.
> 
> 
> [@j0n5m1th](https://github.com/j0n5m1th) Yes, we can swap units with you today. [@kevinwang](https://github.com/kevinwang) [@EthyMoney](https://github.com/EthyMoney) We can swap with you when you're ready as well. [@EthyMoney](https://github.com/EthyMoney) We can send a display module if this is preferred (I believe you are already familiar with the assembly process)
> 
> Thanks for everyone's patience as we worked through this.

Great to hear, now the team can stop joking that it died because I touched it lol

You're correct, we're definitely familiar with getting into already. Feel free to send us just the display module and we can easily swap it out.

Thanks for the update, looking forward to more testing!

brettbbonner, 2025-07-03:
> * The display will automatically dim and shut down after 10 seconds of no-motion.
> * The display will turn on again if you tap your Systemcore.

Are these changes a short-term fix to avoid/reduce failures while the hardware root-cause is investigated, or envisioned as a longer term fix?  If the latter, I have some concerns:
* Robots do endure long sessions when they're on and moving (hours and hours of continuous drive practice, autonomous development, etc.) so this fix doesn't guarantee a short "on" duty-cycle; if having the display on for long periods is a problem, it'll still be an issue.
* The display is going to be a valuable source of information for a CSA/FTA to get quick status - having it turn off after 10 seconds seems like it goes counter to that purpose.  (Cue montage of people in brightly colored hats briskly whacking on-field robots to wake the displays up.)

Enzo-Coutinho, 2025-07-16:
Hello, team 1156 here.

OS Version: Alpha 3.

We left the controller powered on approximately 6~8 hours, and when we checked it again, the screen was off and the red status LED was blinking slowly, indicating a display fault. The Web UI confirms this with multiple display fault messages.

What we tried:
- Power cycle.
- Upgrade to Alpha 4.

However, none of these attempts got the display working again. The rest of the functionalities also seem to be working without any issues.

In the Kernel information, it was also possible to find a warning about a firmware failure on the rpi1.  I'm not sure if it's related to the display issue.

Please let me know if you need any additional information.

How should we proceed with this?
____

https://github.com/user-attachments/assets/4406b3e3-6678-42c0-ae84-4e88563e1e10

[image]

[image]

Enzo-Coutinho, 2025-07-20:
Yes, that would be great, we can make the swap here, thank you.

I'm not exactly sure what may have contributed to the issue, but when we perform the disassembly, I’ll check to see if I find any irregularities with the screen.

If you need any additional information regarding the display, just let us know and we’ll do our best to provide it.