wpilibsuite/SystemcoreTesting #31: USB Storage Drive Support

jasondaming, 2025-06-25:
I tried testing a couple of different storage drives.  The 64gb one formatted as exFAT was not recognized and the 32gb one formatted as Fat32 was.  What is the expected behavior and what drives and formats are supported?

Does it make sense to have a eject button on the USB devices section of the System Tab?  I realize this wasn't supported by the roboRIO and isn't a required just a nice to have.  I also know that I could probably do an unmount manually but that isn't something a typical user would know.

jwbonner, 2025-06-25:
I've also observed USB drive mounting being inconsistent. I've tried a 16GB drive formatted with both FAT32 and exFAT. I believe it consistently claims to be mounted when running `df`, but the mount appears to be functional maybe half of the time (i.e. the contents of the drive actually appear). Unplugging and reconnecting the same drive will often cause different behavior.

jlmcmchl, 2025-06-26:
We plugged a 32Gb usb drive into systemcore yesterday to collect datalogs and found that `robot.service` could not write to `/U/logs`, so logs instead accumulated within `/home/systemcore/logs`. I noticed that `/U` was mounted with root ownership, but given that robot code appears to be running as root I'm unclear as to why logs were recorded to internal storage.

jasondaming, 2025-06-30:
I was just hotplugging.  I didn't think to test on boot up but am happy to report that both devices now show up on boot up (I am pretty sure I tested that in my testing on the previous version).

Unfortunately once unplugged I can't get the device back without rebooting.

jasondaming, 2025-07-01:
Hopefully this helps

[dmesg_output.txt](https://github.com/user-attachments/files/21005213/dmesg_output.txt)

jasondaming, 2025-07-02:
@bhjelstrom 
I am really unsure of the state my systemcore is in.  I updated to 159.  I now see a higher voltage.  However I don't see any of the UI improvements like the eject USB.  Should we troubleshoot this more or should I just go to 160 and see if that resolves things?

jasondaming, 2025-07-02:
Some USB devices when I plug into windows I get a Scan/Fix this device dialog but when I plug in both of these they immediately pop open with no warning to fix.

jasondaming, 2025-07-02:
I don't always get the extra device.  I just rebooted and pluged 1 in and didn't get it.

bhjelstrom, 2026-10-13:
Improved in Beta units, fixed in SoCal + production units

closed 2026-10-13 by bhjelstrom