wpilibsuite/SystemcoreTesting #41: Chrome reports high resource usage for SYSTEMCORE Tab

gerth2, 2025-06-26:
[image]

While doing some browsing with the tab up, I get this warning from chrome. First time I've seen it. I had the system tab up, so presumably plots and the 3d visualization were involved?

[Systemcore_gerth_Jun25_Trace-20250625T195126.json](https://github.com/user-attachments/files/20914624/Systemcore_gerth_Jun25_Trace-20250625T195126.json)

Here's a quick chrome profile of the tab on my PC

codydg, 2025-06-29:
Our preliminary testing is showing very high CPU usage when on the System tab with all the charts and the 3D render up.

With that tab open, our CPU is sitting at 100% usage, with Chrome being 65% of it (Not much else running but Driver Station does take a notable 10%). The "Desktop Window Manager" process is also using 10% when we view this tab.

When we close the tab, Chrome's usage drops to 15%, and Desktop Window Manager drops to 3%.

codydg, 2025-07-02:
My experience was on Alpha 1. If I update the SystemCore firmware to Alpha 2, do I need to update anything else with it?

gerth2, 2025-07-03:
originally reproduced on alpha 1. No issues seen yet on alpha 2, will check again with 3 soon

codydg, 2025-07-06:
@bhjelstrom 
There wasn't very much output - not sure if I'm missing something or if this is all you're looking for!

The computer that struggled has an Intel i5-1035G1 with 32GB RAM. Here's the console after refreshing, with Verbose logging checked:

```
GLOBALS LOADED
main.ab0a8c05.js:2 HOME
main.ab0a8c05.js:2 IDX 1
main.ab0a8c05.js:2 Attempting reconnection
main.ab0a8c05.js:2 ws://robot.local:4800
main.ab0a8c05.js:2 Attempting reconnection
main.ab0a8c05.js:2 Header websocket connection established
main.ab0a8c05.js:2 Subscribed to minimal data stream
main.ab0a8c05.js:2 WebSocket connection established HW
```

codydg, 2025-08-08:
I went ahead and re-did this on the same laptop as before with Alpha 5, and had no noticeable issues.

According to Task Manager...
CPU usage of Google Chrome's process was fairly consistently at ~31-35%
GPU usage of Intel(R) UHD Graphics was at ~11% (While the GPU usage is not specific to Chrome, it was 0-1% prior to opening that web page)

The computer also has 32GB of RAM so I'm sure that's not the limiting factor here.
```
GLOBALS LOADED
main.2c02c9c6.js:2 HOME
main.2c02c9c6.js:2 IDX 0
main.2c02c9c6.js:2 Attempting reconnection
main.2c02c9c6.js:2 Header websocket connection established
main.2c02c9c6.js:2 Subscribed to minimal data stream
main.2c02c9c6.js:2 HOME
main.2c02c9c6.js:2 IDX 1
main.2c02c9c6.js:2 WEBGL SUPPORT: true
main.2c02c9c6.js:2 GRAPHICS: ANGLE (Intel, Intel(R) UHD Graphics (0x00008A56) Direct3D11 vs_5_0 ps_5_0, D3D11)
main.2c02c9c6.js:2 CPU THREADS: 8
main.2c02c9c6.js:2 3D ENABLED true
main.2c02c9c6.js:2 ws://robot.local:4800
main.2c02c9c6.js:2 Attempting reconnection
main.2c02c9c6.js:2 HOME
main.2c02c9c6.js:2 WebSocket connection established HW
main.2c02c9c6.js:2 Subscribed to full data stream
```