wpilibsuite/SystemcoreTesting #49: Robot program running as root?

ThadHouse, 2025-06-26:
### Discussed in https://github.com/wpilibsuite/SystemCoreTesting/discussions/36

<div type='discussions-op-text'>

<sup>Originally posted by **jwbonner** June 25, 2025</sup>
It looks like the robot service currently runs the user program as root. Is there a good reason to do this instead of running as the `systemcore` user (with appropriate permissions for any resources it needs to access)? It seems like running as root increases the risk that a misbehaving user program could cause major issues on the device (for example, suppose a user program with an incorrect path configured somewhere starts inadvertently writing/overwriting files in a system directory). I assumed that this issue was the reason for running the user program as `lvuser` on the RIO.

One related issue regarding permissions is that it looks like `systemcore` doesn't have permission to write to USB drives with the current auto-mounting scheme. Obviously if the user program ran as that user it would need to be configured with write access to USB drives.</div>

Kevin-OConnor, 2025-06-26:
To do this we need to:

- Update the service file to specify User=systemcore in the Service section
- Update the service file to set LimitRTPRIO in the Service section to whatever value we want robot programs to be able to utilize. If we want the CAN thread to remain the highest priority, this should be LimitRTPRIO=50
- Ensure the systemcore user is able to write to all desired locations (such as USB drives)

rzblue, 2025-07-05:
> If we want the CAN thread to remain the highest priority, this should be LimitRTPRIO=50

High priority kernel threads (including interrupt handlers) also run with priority 50 so this would be a reasonable limit for user code.

auscompgeek, 2025-08-01:
> Ensure the systemcore user is able to write to all desired locations (such as USB drives)

Other notable things that robot programs need access to:

- CAN buses (hopefully socketcan makes this a non-issue?)
- I2C buses
- V4L devices (i.e. webcams)
- serial ports

AustinSchuh, 2025-08-01:
We need to make sure `systemcore` is in the right groups for those (`video`, `tty`, etc), but that should flow correctly from there.

For CAN busses and i2c busses, we might need some udev rules to make sure the names are stable.  Otherwise we are relying on enumeration order which isn't guaranteed to be stable.