wpilibsuite/SystemcoreTesting #57: Using DutyCycleEncoder causes crash on `.get`

pjreiniger, 2025-06-27:
We have a REV Through Bore Encoder that we wired into the roborio since we didn't have a Vortex breakout board. When using it with system core the robot code crashes when calling `.get()`

It gets constructed fine, and the SmartIO web ui shows that it got auto configured to be a PWM input and displayed seemingly sensical values when we rotated our pivot, but we could not read from the sensor in java.

```
7:49:35.658 PM
ERROR  1  The startCompetition() method (or methods called by it) should have handled the exception above.  edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:318) Warning  1  The robot program quit unexpectedly. This is usually due to a code error.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information.  edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:311) ERROR  1  Unhandled exception: edu.wpi.first.hal.util.HalHandleException:  Code: -1098. HAL: A handle parameter was passed incorrectly  com.gos.reefscape.subsystems.PivotSubsystem.getAbsoluteAngle(PivotSubsystem.java:184) 

```

ThadHouse, 2025-06-27:
https://docs.wpilib.org/en/2027/docs/yearly-overview/known-issues.html#dutycycle-getoutput-crashes

Noted on the Known Issues page.

rzblue, 2025-07-15:
This is fixed by https://github.com/wpilibsuite/allwpilib/pull/8076

closed 2025-07-28 by Kevin-OConnor