wpilibsuite/SystemcoreTesting #58: Photelectric sensor does not read boolean values correctly

pjreiniger, 2025-06-27:
This might just be because the sensor isn't happy working on 3.3v instead of 5. I don't have the part number,  but we used a photoelectric sensor like [this ](https://www.amazon.com/Photoelectric-Switch-Sensor-E18-B03P1-5-30cm/dp/B07SW6JF6S)

When we try to use it as a `DigitalInput` like we did on the roborio, the indicator light on the sensor will correctly change, but the value in code does not. I tried changing it to an analog input to see what kind of voltage it was spitting out and got the following values

```
   Low - `.getVoltage()` ~0.037, ~46 in systemcore gui
   High - `getVoltage()` ~1.12, ~1400 in systemcore gui
```

ThadHouse, 2025-06-27:
SystemCore has pull downs instead of pull ups on the IO. So if it worked on the roborio, its possible you'll need to add an extrnal pull up resistor.

gerth2, 2025-06-27:
{subscribed}

We have two of those same ones on our robot too - will update here if/when we get to testing them. I have a spare sitting around I might snag for playing with over the weekend.