wpilibsuite/SystemcoreTesting #70: Toolchain is excessively large

Gold856, 2025-06-29:
There are a lot of unnecessary files included in the toolchain. One of the most egregious examples is the Go cache:
[image]
One gigabyte out of 3.6 is spent on Go caches. 0.9 gigabytes is spent on various Go binaries under systemcore-aarch64-toolchain/lib. I'm not exactly sure how much is needed to build the image, but it definitely looks like there's opportunities to slim it down significantly.

ThadHouse, 2025-06-29:
> Also, it seems there are actually two copies of the JVM? `/usr/lib/jvm` and `/usr/local/bin/llcodeserver/data/extensions/redhat.java/extension/jre/17.0.13-linux-aarch64`
The 2nd one comes with the vscode extension and is integrated into it. There’s nothing we can do about that one.

rzblue, 2025-06-29:
> jdk has insane dependencies that aren't actually necessary for our purposes, etc.

After also encountering this when building an image with buildroot, I was able to bundle the generic aarch64 Adoptium JDK build which has been working well.

Another thing that will help reduce the on-device filesystem size (though not the toolchain size) would be to enable module xz compression in the kernel build, which reduce the size of the modules by about 70%.

Gold856, 2025-07-02:
The size is definitely a lot better with release 161, and the unpack time seems a lot better as well. Still seems like the above points apply w.r.t to optimizing the Python stuff, using the Java extension without a bundled JRE, and bundling a smaller JDK without the unnecessary dependencies.

bhjelstrom, 2026-10-14:
vision python scripts using the system opencv is one use case. There's a total of about 300MB we can save from the compressed bundle removing GO. This will be evaluated in release 17 or 18.

closed 2026-10-14 by bhjelstrom

Gold856, 2026-10-30:
17 and 18 are out and this has not been resolved.

auscompgeek, 2026-10-31:
> python scripts using the system opencv is one use case

I'm a bit confused by this statement. Are we expecting that teams will want to build their own CPython extensions that link against the system OpenCV libraries? If teams are doing that, does it still make sense to be writing Python at that point?