wpilibsuite/SystemcoreTesting #71: Mesa not present in the OS image

qwertychouskie, 2025-06-29:
Poking around the OS image, it seems Mesa is completely absent, which means no OpenCL, which means no GPU-accelerated OpenCV, which means leaving free vision performance on the table.  Mesa and related utilities (`glxinfo`, `eglinfo`, `vulkaninfo`, `clinfo`, `clpeak`) should be included in the OS image.

rzblue, 2025-06-29:
It doesn't look like there's an opencl driver for the VideoCore VII gpu.

qwertychouskie, 2025-06-29:
> It doesn't look like there's an OpenCL driver for the VideoCore VII gpu.

Rusticl (part of Mesa) provides OpenCL.

qwertychouskie, 2025-07-17:
> For now I think a docker-based package is the right tool for something like this.

OpenCV automatically uses OpenCL when available to GPU-accelerate parts of the vision pipeline.  Putting Mesa in a Docker image would defeat the purpose of having it.