wpilibsuite/SystemcoreTesting #73: CAN bus utilization display underreports utilization

rzblue, 2025-06-29:
The utilization display in the web interface appears to be using the interface statistics from the kernel, which only tracks the size of the frame payload, not the full message as transmitted on the bus (with message ID, CRC, stuff bits, etc). As an example, on my system, `canbusload can_s0@1000000 -ertb` (exact bitstuffing calculation enabled) is reporting a utilization of about 61% on can_s0, while the web ui only shows about 25%.

[image]

[image]

Kevin-OConnor, 2025-07-28:
@bhjelstrom I see this note in OS Alpha 5 (166): Increase accuracy of can bus load calculation

Does that address this, or is there further work needed?

bhjelstrom, 2025-12-15:
Should be a lot better now. Using the algorithms from can_utils in release 9.