wpilibsuite/SystemcoreTesting #77: Support internet bridging

qwertychouskie, 2025-06-30:
(Copied from https://github.com/wpilibsuite/allwpilib/issues/7928, as this repo seems more applicable)

**Is your feature request related to a problem? Please describe.**
When you have the computer connected to the robot to code, not having an internet connection is a massive pain.  Also, if teams are using onbot, having the ability to pull from and push to GitHub and such is a huge benefit.

**Describe the solution you'd like**
Provide a nice UI for enabling bridging, so when you are connected to the robot, you also have internet connectivity.  The RPi hardware supports this (https://raspberrypi.stackexchange.com/a/89804), so it should theoretically just be a matter of having an easy way for teams to enable this.

Here's what it might look like:
```
                 wifi                         wifi uplink         wan
 ds-&-laptop <~.~.~.~.~> (ap@wlan0)RPi(wlan0) <.~.~.~.~.> router <───> INTERNET
            ╲             ╱               ╲
           (dhcp    192.168.43.1         (dhcp
         from RPi)                    from router)

```

**Describe alternatives you've considered**
A second USB WiFI adapter plugged into your computer, but that's extra hardware that is not needed, and also Macs don't really support that setup.

**Additional context**
Was a Discord convo, putting the information I found here so it doesn't get lost/forgotten.

dellagd, 2025-06-30:
Trying to understand... is the suggestion here is to support having the SystemCore connect to, say, the normal internet WiFi available in your practice space through the SystemCore WiFi wlan0 interface (as a client), which would then allow:

1. SystemCore to access the internet itself (pretty straightforward)
2. Any device connected to the Vivid robot radio (router) to access the internet using the SystemCore as a gateway?

The other element of configuration for this would be the Vivid radio advertising that the SystemCore is a valid upstream gateway - unless we're moving to a architecture where the RPi and not the Vivid radio runs the DHCP server?

ThadHouse, 2025-06-30:
> Trying to understand... is the suggestion here is to support having the SystemCore connect to, say, the normal internet WiFi available in your practice space through the SystemCore WiFi wlan0 interface (as a client), which would then allow:
> 
> 1. SystemCore to access the internet itself (pretty straightforward)
> 2. Any device connected to the Vivid robot radio (router) to access the internet using the SystemCore as a gateway?
> 
> The other element of configuration for this would be the Vivid radio advertising that the SystemCore is a valid upstream gateway - unless we're moving to a architecture where the RPi and not the Vivid radio runs the DHCP server?
> 
> Regardless, I would also love to have this built-in and supported.

FTC won't be using the Vivid Radio, and will just be hooked up to the internal WiFi on systemcore. So it would be making a bridge so a laptop could be connected to the robot, and then the robot would be connected to WiFi as well.

bhjelstrom, 2026-10-05:
We will not support this functionality

closed 2026-10-05 by bhjelstrom