wpilibsuite/SystemcoreTesting #92: AdvantageScope lite browser installation fails

jlmcmchl, 2025-07-03:
I had initially written this in the AdvantageScope repo, but I was able to install the package after using rsync to send the ipk file over using `opkg install advantagescope.ipk`.

Tested on os release 161.

**Describe the bug**
When I upload the ipk file to the systemcore, AdvantageScope Lite fails to install. The 'loading' gif runs for a few seconds, and then promptly disappears.

Additionally, no logs are presented regarding installation, like one receives after a successful install.

**To Reproduce**
Steps to reproduce the behavior:

1. Install Systemcore os Alpha 3 onto a systemcore unit
2. Navigate to http://robot.local
3. Drag + Drop the AdvantageScope Lite v27.0.0-alpha-1 ipk file install prompt
4. Observe described behavior

**Expected behavior**
AdvantageScope Lite installs

**Additional context**
There is nothing written to the browser's console during this operation.
Logs:
```
$ sudo !!
sudo opkg install advantagescope-lite-v27.0.0-alpha-1.ipk
Installing advantagescope-lite (27.0.0-alpha) on root.
Configuring advantagescope-lite.
Configuring service for advantagescope-lite
Enabling service for auto-start on boot...
Created symlink /etc/systemd/system/multi-user.target.wants/advantagescope-lite.service → /etc/systemd/system/advantagescope-lite.service.
Starting service...
Service started successfully
Service is now running
Package installation complete
```

jwbonner, 2025-07-03:
Looks like the error is about the file size. Did this limit change?

```
[PKG_MGR_ERROR] UPLOAD: Upload error: File too large (max 50MB)
```

jlmcmchl, 2025-07-15:
I had this issue again with the Alpha 4 image - using rsync to copy the ascope lite image onto the device worked fine, but installing via the browser failed.