wpilibsuite/SystemcoreTesting #97: Sufficiently Wonky TCP traffic causes reset

gerth2, 2025-07-06:
Not a huge priority but just a fun discovery. While fiddling around with [clumsy](https://jagt.github.io/clumsy/) to help simulate lag/drops in network on field, I attempted their `Tamper` option (randomly nudges bits in packets) and was able to get the whole SystemCore to reset pretty consistently.

[image]

wireshark capture of the event: https://discord.com/channels/176186766946992128/1365422855378178048/1391236907085729854

I can get it to happen very reliably on Alpha 3 - any specific onboard logs to snag to start debug?

gerth2, 2025-07-06:
Two quick samples, using the same "preturb packets" technique on a Pi3 flashed with photonvision does not seem to restart:

https://github.com/user-attachments/assets/2e9b506f-4075-41c5-a8e4-3d4fa9c65e2d

Doing it to a roboRIO with 2025 firmware has mixed results - sometimes just messing up communication, other times actually resetting:

https://github.com/user-attachments/assets/05c45d21-e256-4229-aa01-ee2702c1d005

rzblue, 2025-07-06:
> any specific onboard logs to snag to start debug?

Checking the tail end of the journal from the previous boot (`journalctl -b -1`) after it restarts would be a good starting point

rzblue, 2025-07-06:
yep, forgot #88 hasn't been fixed yet

qwertychouskie, 2025-07-06:
I feel like robot control packets should maybe have some sort of CRC so that bitflipped packets can't cause problematic behavior.  If a controller input value bitflipped and that packet wasn't dropped as invalid, that could cause real damage to robot and/or human.

ThadHouse, 2025-07-06:
We made in internal issue to add a CRC to the new protocol version. The existing NI ds doesn’t have one though.