wpilibsuite/SystemcoreTesting #101: Web UI System tab crashes on Alpha 3

mjansen4857, 2025-07-07:
Just flashed and deployed our code project, the system page crashes and shows only a white screen.

Browser console output:
```
WebSocket connection closed HW [main.ab0a8c05.js:2:3083046](http://robot.local/static/js/main.ab0a8c05.js)
GLOBALS LOADED [main.ab0a8c05.js:2:554112](http://robot.local/static/js/main.ab0a8c05.js)
HOME [main.ab0a8c05.js:2:3112111](http://robot.local/static/js/main.ab0a8c05.js)
IDX 1 [main.ab0a8c05.js:2:3109573](http://robot.local/static/js/main.ab0a8c05.js)
Attempting reconnection [main.ab0a8c05.js:2:678659](http://robot.local/static/js/main.ab0a8c05.js)
ws://robot.local:4800 [main.ab0a8c05.js:2:3082599](http://robot.local/static/js/main.ab0a8c05.js)
Attempting reconnection [main.ab0a8c05.js:2:3082748](http://robot.local/static/js/main.ab0a8c05.js)
Header websocket connection established [main.ab0a8c05.js:2:678753](http://robot.local/static/js/main.ab0a8c05.js)
Subscribed to minimal data stream [main.ab0a8c05.js:2:678987](http://robot.local/static/js/main.ab0a8c05.js)
WebSocket connection established HW [main.ab0a8c05.js:2:3082842](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: Error creating WebGL context. [main.ab0a8c05.js:2:1340974](http://robot.local/static/js/main.ab0a8c05.js)
HOME [main.ab0a8c05.js:2:3112111](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) 2 [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) 2 [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
Failed to create WebGL context: WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) [main.ab0a8c05.js:2:1340160](http://robot.local/static/js/main.ab0a8c05.js)
THREE.WebGLRenderer: A WebGL context could not be created. Reason:  WebGL creation failed: 
* tryANGLE (FEATURE_FAILURE_EGL_NO_CONFIG)
* Exhausted GL driver options. (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS) 2 [main.ab0a8c05.js:2:1341926](http://robot.local/static/js/main.ab0a8c05.js)
[... 54 lines cut ...]
    kl http://robot.local/static/js/main.ab0a8c05.js:2
    bu http://robot.local/static/js/main.ab0a8c05.js:2
    gu http://robot.local/static/js/main.ab0a8c05.js:2
    mu http://robot.local/static/js/main.ab0a8c05.js:2
    au http://robot.local/static/js/main.ab0a8c05.js:2
    iu http://robot.local/static/js/main.ab0a8c05.js:2
    S http://robot.local/static/js/main.ab0a8c05.js:2
    P http://robot.local/static/js/main.ab0a8c05.js:2
    1467 http://robot.local/static/js/main.ab0a8c05.js:2
    n http://robot.local/static/js/main.ab0a8c05.js:2
    7108 http://robot.local/static/js/main.ab0a8c05.js:2
    n http://robot.local/static/js/main.ab0a8c05.js:2
    534 http://robot.local/static/js/main.ab0a8c05.js:2
    n http://robot.local/static/js/main.ab0a8c05.js:2
    1168 http://robot.local/static/js/main.ab0a8c05.js:2
    n http://robot.local/static/js/main.ab0a8c05.js:2
    <anonymous> http://robot.local/static/js/main.ab0a8c05.js:2
    <anonymous> http://robot.local/static/js/main.ab0a8c05.js:2
    <anonymous> http://robot.local/static/js/main.ab0a8c05.js:2
[main.ab0a8c05.js:2:1340785](http://robot.local/static/js/main.ab0a8c05.js)
Cleaning up header WebSocket connection [main.ab0a8c05.js:2:679553](http://robot.local/static/js/main.ab0a8c05.js)
end controller [main.ab0a8c05.js:2:3083217](http://robot.local/static/js/main.ab0a8c05.js)
WebSocket connection closed [main.ab0a8c05.js:2:679361](http://robot.local/static/js/main.ab0a8c05.js)
WebSocket connection closed HW
```

ThadHouse, 2025-07-07:
This is looking like an error on your computer side, where its failing to load opengl in your browser. What OS and browser are you using to connect?

mjansen4857, 2025-07-08:
Its working for us on Firefox now as well. Maybe it updated? Regardless, I agree just disabling the visualizer instead of the whole page crashing is a nice solution.