wpilibsuite/SystemcoreTesting #98: Debug symbols for system libraries

PeterJohnson, 2025-07-07:
For remote debugging, it would be very useful to have debug symbols included in the toolchain sysroot (or separately packaged).  It’s not necessary to include them in the image.  Split debug info (eg .so.debug or similar) is the right way to do this.

Debug symbols should probably be excluded for closed source components like mrccomm, although building them (with restricted access to the resulting files) would be good to do.

qwertychouskie, 2025-07-07:
It'd be best to set up debuginfod, so GDB will offer to automatically download symbols when you start debugging a binary.  debuginfod makes the experience soooooooo much better.

PeterJohnson, 2025-07-07:
That may not be practical, we'd have to find a way to host it, and it requires Internet access simultaneous with device access.

qwertychouskie, 2025-07-07:
> That may not be practical, we'd have to find a way to host it, and it requires Internet access simultaneous with device access.

https://github.com/wpilibsuite/SystemCoreTesting/issues/77 covers internet access.  As for hosting, there are definitely options (when setting up clipfarm we found that Cloudflare basically gives away tons of bandwidth for free lol)

PeterJohnson, 2025-07-08:
That doesn’t help at events, at labs without internet, etc.  We must have a local option available.

PeterJohnson, 2025-07-09:
Is debuginfod available and buildable on Mac and Windows?  If it means having a background service installed and running, that’s also problematic unless it’s very automatic.

Gold856, 2025-07-19:
Should this be closed? It seems like the toolchain has debug info for a bunch of libraries.