wpilibsuite/SystemcoreTesting #116: Disabling the robot can leave digital outputs high

desertgreg, 2025-07-12:
I wrote a simple robot program which uses a DigitalOutput to toggle an io pin on and off.  Depending on when I hit the 'Disable' button the io pin will be left in either high or low.

PeterJohnson, 2025-07-12:
Is it possible to control the outputs in disabled mode in general?

desertgreg, 2025-07-12:
I don't think so. I added code to toggle all of the digital outputs to DisabledPeriodic but they are not toggling in disabled mode.  I've attached code to repro it, just rename .txt to .java.  The steps are to run the program, enable, observe the digital output pins toggling once every update (50ms) then disable.  I can get it to leave the pins high sometimes and low sometimes depending on when it gets disabled.

[Robot.txt](https://github.com/user-attachments/files/21199711/Robot.txt)