wpilibsuite/SystemcoreTesting #129: UpDownCounter's `.reset` not currently resetting

gerth2, 2025-07-19:
MRE:

```java

public class EncoderTest {

    UpDownCounter udc;

    public EncoderTest(){
        udc = new UpDownCounter(0, EdgeConfiguration.kRisingEdge);
    }

    public void update(){
        SmartDashboard.putNumber("EncoderTest/SystemCoreCount", udc.getCount());
    }

    public void reset(){
        udc.reset()
    }
    
}

```

Call `update()` in robotPeriodic(), and `reset()` in `autonomousInit()`.

Apply some pulses to pin 0. You should see counts go up in NT. Then enable in autonomous (Causing `reset()` to get called). 

I'd expect `getCount()` to start returning 0, but it keeps returning its old non-zero value.

Kevin-OConnor, 2025-07-28:
@bhjelstrom currently entering a value into the valset key for an IO pin configured as a counter has no effect. Do we want to implement setting or resetting at that layer so it will reflect properly in the web interface or just handle it at the HAL layer with a stored offset?

bhjelstrom, 2025-12-16:
I take it back, this should be a soft offset in wpilib. Thanks!

ThadHouse, 2025-12-16:
Does it at least clear when changing modes? If so that is fine. Otherwise I'd have some concerns about jumping values when code restarts.