wpilibsuite/SystemcoreTesting #131: Feature: Web UI for periodic CAN frames

jwbonner, 2025-07-21:
A nice features of the REV Hardware Client is the interface below for viewing CAN data, which is accessible when connected via USB to any REV device on the CAN bus. It's a simple printout of the recent CAN frames detected on the bus with a few key fields decoded. Each message ID has its own row, which disappears if it isn't seen for a few seconds.

We've found that this is an invaluable tool for allowing students to "peek behind the curtain" on CAN-based devices, since it's very easy to see how (for example) moving an encoder causes the associated CAN frame to update. For the majority of vendors that publish CAN specs, students can even use this data to manually decode a few frames and see how the data is actually communicated. We find that there are a lot of problems (both electrical and software) that get easier for students to grasp when they have some context for what's happening under the hood (e.g. "Why do we care about periodic status frequencies?").

In addition to the learning experience, there are also plenty of ways that this can be helpful for debugging. It provides a unified interface to see which devices are active on the bus without needing to switch between multiple vendor tools, and we can confirm which frames are being published (again, for the majority of vendors who document that information).

As a future enhancement, it would be very nice to make an interface like this accessible through the SystemCore web UI without needing to rely on a separate tool. I think the basics seen here are really the most critical, but there are plenty of enhancements I could imagine adding too. Timestamps? Frequencies for each row? Utilities for decoding common numeric types?

[image]

Kevin-OConnor, 2025-07-23:
I believe LL has a version of this that just missed image Alpha 5 in an effort to get an image with LED support and some other fixes out to enable a WPILib Alpha 2

bhjelstrom, 2025-12-15:
https://www.youtube.com/watch?v=3iXizaWo7wE

In the latest release.

The Can bus monitor supports the REV json spec and DBC for decoding.