wpilibsuite/SystemcoreTesting #143: `robotpy deploy --debug` failing on chmod

gerth2, 2025-07-25:
MRE:

Install robotpy

new folder

`robotpy init`

fill out default `robot.py` code:

```py
import wpilib

class MyRobot(wpilib.TimedRobot):
    def robotInit(self):
        pass
```

run `robotpy deploy --debug`, allow test creation and run (should pass), deploy fails with the following messages:

```

19:40:54:580 INFO    : deploy              : Robot project requirements:
19:40:54:581 INFO    : deploy              : - robotpy==2027.0.0a2
19:40:54:925 INFO    : robotpy.installer   : RobotPy Installer 2027.0.0a2
19:40:54:925 INFO    : robotpy.installer   : -> caching files at C:\Users\chris\wpilib\2027\robotpy
Robot setup (hit enter for default value):
Team number or robot hostname: 1736
19:40:57:858 INFO    : robotpy.installer   : Settings stored at C:\Users\chris\git\SystemCoreRobotpyDebugTest\.wpilib\wpilib_preferences.json
19:40:57:881 INFO    : robotpy.installer   : Connecting to robot via SSH at 10.17.36.2
19:40:57:893 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_9.6)
19:40:57:989 INFO    : paramiko.transport  : Auth banner: b'Warning: your password will expire in 0 days.\n'
19:40:58:047 INFO    : paramiko.transport  : Authentication (password) successful!
19:40:58:066 INFO    : robotpy.installer   : -> SystemCore disk usage 1.9G/4.5G (45% full)
19:40:58:127 INFO    : robotpy.installer   : -> SystemCore memory 193.4M/4087.0M (95% full)
19:40:58:127 INFO    : robotpy.installer   : Clearing FRC C++/Java user programs if present
19:40:58:254 INFO    : paramiko.transport.sftp: [chan 3] Opened sftp connection (server version 3)
19:40:58:260 INFO    : paramiko.transport.sftp: [chan 3] sftp session closed.
19:40:58:380 INFO    : paramiko.transport.sftp: [chan 5] Opened sftp connection (server version 3)
19:40:58:387 INFO    : paramiko.transport.sftp: [chan 5] sftp session closed.
19:40:58:552 INFO    : deploy              : All project requirements already installed
19:40:58:635 INFO    : deploy              : Killing robot program
ERROR: configuring command: Command 'chmod +x /home/systemcore/robotCommand' returned non-zero error status 1
PS C:\Users\chris\git\SystemCoreRobotpyDebugTest>
```

gerth2, 2025-07-25:
Running on alpha4 release of systemcore image

virtuald, 2025-07-25:
Did you previously have Java installed on this?

gerth2, 2025-07-25:
> Did you previously have Java installed on this?

Yes

closed 2025-07-25 by gerth2