wpilibsuite/SystemcoreTesting #145: HAL Runtime Error on calling `RobotController` `isBrownedOut()`

gerth2, 2025-07-25:
Replicated with python, but likely not python specific.

```py
from wpilib import RobotController
#...
tmp = RobotController.isBrownedOut()
```

produces:

```
Exception has occurred: RuntimeError
HAL: A handle parameter was passed incorrectly: IsBrownedOut
  File "\home\systemcore\py\utils\rioMonitor.py", line 110, in _updateVoltages
    if not RobotController.isBrownedOut():
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "\home\systemcore\py\utils\rioMonitor.py", line 65, in _updateFast
    self._updateVoltages()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "\home\systemcore\.python\lib\python3.13\threading.py", line 994, in run
    self._target(*self._args, **self._kwargs)
  File "\home\systemcore\.python\lib\python3.13\threading.py", line 1043, in _bootstrap_inner
    self.run()
  File "\home\systemcore\.python\lib\python3.13\threading.py", line 1014, in _bootstrap
    self._bootstrap_inner()
RuntimeError: HAL: A handle parameter was passed incorrectly: IsBrownedOut
```

gerth2, 2025-07-25:
Code runs fine in simulation, but produces HAL errors on SystemCore only

gerth2, 2025-07-25:
Just found same issue on RobotController.getEnabled3V3() as well:

Exception has occurred: RuntimeError
HAL: A handle parameter was passed incorrectly: GetEnabled3V3
  File "\home\systemcore\py\utils\rioMonitor.py", line 111, in _updateVoltages
    self.railFault3p3v.set(not RobotController.getEnabled3V3())
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "\home\systemcore\py\utils\rioMonitor.py", line 65, in _updateFast
    self._updateVoltages()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "\home\systemcore\.python\lib\python3.13\threading.py", line 994, in run
    self._target(*self._args, **self._kwargs)
  File "\home\systemcore\.python\lib\python3.13\threading.py", line 1043, in _bootstrap_inner
    self.run()
  File "\home\systemcore\.python\lib\python3.13\threading.py", line 1014, in _bootstrap
    self._bootstrap_inner()
RuntimeError: HAL: A handle parameter was passed incorrectly: GetEnabled3V3

gerth2, 2025-07-25:
> This isn’t implemented yet on SC.

All good, easy to comment out for now, and nothing critical to controls logic yet.

I'll keep chunking through our repo and just comment out things with the HAL parameter error (looks like there might be one or two more).

jonahsnider, 2025-11-20:
Is there any way to test this behavior in sim? Or a list of what APIs are off-limits for now?