wpilibsuite/SystemcoreTesting #148: Limelight Hardware Manager Drivers indicator and installation

chauser, 2025-07-27:
The Drivers indicator at the top of the Flash OS tab of the Limelight Hardware Manager is green immediately after installing the program, but Select Target Device is unable to find any targets. Clicking the Install/Reinstall drivers button installs the drivers and then the target device can be found -- and the indicator remains green. Given the amount of time that it takes to install the drivers and the speed at which the program installation goes, it seems likely that the drivers are not installed when the program is installed and that the initial green display of the Drivers indicator is wrong.

Relatedly, I think it would be good to set expectations for how long the driver installation takes and that there are four very similar-looking steps that all need to complete. I didn't time it, but my rough estimate is that each of those four steps takes 20-30 seconds. I wasn't prepared for that, didn't realize there were 4 drivers, and actually quit the program thinking it had hung in a loop.

bhjelstrom, 2025-12-10:
Pt1: [image]

bhjelstrom, 2025-12-10:
+ More comprehensive driver check

[image]

With a warning message if the driver check fails on a locked down computer.

[image]