wpilibsuite/SystemcoreTesting #151: Advantage kit auto log doesnt work.

RomHollander, 2025-07-27:
we are trying to use AutoLog and the its build  scssful but there are some Erorrs on the inputs and AutoLog import 
we have the Vendordeps of Adventage kit

jwbonner, 2025-07-28:
The most likely issue is that the annotation processor for AdvantageKit isn't installed in the `build.gradle` file. In addition to adding the vendordep JSON file, did you add the code to the `build.gradle` file as shown in the [installation docs](https://docs.advantagekit.org/getting-started/installation/existing-projects)?

jwbonner, 2025-08-06:
@KlausKollerKroeff The AutoLog annotation appears to be working correctly in that repo. Are you seeing issues when building with Gradle or only in VSCode?

KlausKollerKroeff, 2025-08-06:
I'm not seeing any issues when deploying the code, it is related to not seeing the AutoLog input values in Advantage Scope. When I try running the robot code, any of those variables seem to appear.

KlausKollerKroeff, 2025-08-07:
Thanks for the tip, I just resolved the issue, I had missed calling that method in the `periodic()` method of the `BaseSwerveSubsystem` class. Everything started showing up correctly in Advantage Scope. Appreciate the help!

closed 2025-12-09 by bhjelstrom