wpilibsuite/SystemcoreTesting #156: Attempting to deploy code with full storage deletes current robot program

mjansen4857, 2025-08-03:
When attempting to deploy code when the internal storage of SystemCore is full, the deployment process will delete the current code, then fail to deploy with a generic SFTP failure error message. This leaves the robot with no code, and no clear indication of why deployment is failing. 

This process can be improved by determining if SystemCore has enough available storage space before deleting the current robot program, then printing a more specific "out of storage space" error instead of a generic SFTP error.

ThadHouse, 2025-08-03:
Just a note, this actualy isn't new behavior, the roborio has the exact same issue. We wish SSH in general resulted in a better error when out of disk space.

rzblue, 2025-08-03:
Now that there will be a good amount of margin, the deploy process could check for some threshold of free space before deploying? That'd also just generally be a good warning for users before it fills up completely and weird behavior ensues.

auscompgeek, 2025-08-03:
To reduce the window where we could result in no robot code deployed, it might be worth considering a blue/green style deploy where the robot code is uploaded, then the the old and new get swapped, and then the old is deleted after that's successful.

We started doing that in RobotPy after my team interrupted a deploy midway and ended up with a dead robot on the field.