wpilibsuite/SystemcoreTesting #157: DriverStation.isEnabled will sometimes blip as false while enabled and connected to FMS

mjansen4857, 2025-08-03:
We noticed at our offseason event yesterday that `DriverStation.isEnabled()` (as well as other methods requiring a connection to the FMS/DriverStaion) will sometimes randomly report an incorrect value for 1 robot loop. It seems that the robot program believes it has disconnected from the FMS and reports those values accordingly, but both the DS logs and the field logs show no actual disconnects. We only saw this happen in two sequential matches, then didn't see it for the rest of the event, so it's pretty rare.

We unfortunately no longer have our robot code logs showing this, as we had to quickly delete them on the field due to #156 

Another mentor theorizes that this might be due to too aggressively checking for the heartbeat from the DS, so the robot disables itself for one loop because it believes it lost connection. I don't know any specifics of how this works, but we are running our robot loop at 100Hz on SystemCore, so maybe it's a side effect of this?

ThadHouse, 2025-08-03:
Did you update to Alpha 2? Alpha 1 likely had a bug here, but that should have been fixed in Alpha 2.

Kevin-OConnor, 2025-08-03:
Were you running WPILib Alpha 1 or Alpha 2?

closed 2025-08-03 by ThadHouse