wpilibsuite/SystemcoreTesting #161: Code fails to deploy in some cases when encrypted ssh key is present in ~/.ssh

rzblue, 2025-08-07:
```
systemcore @ robot.local: Resolved but not connected.
  Reason: RuntimeException
  org.apache.sshd.common.SshException: [ssh-connection]: Failed (IllegalStateException) to execute: Failed (FailedLoginException) to load key from /Users/username/.ssh/id_rsa: No password data for encrypted resource=/Users/username/.ssh/id_rsa
```
Reported to me, currently unsure what makes it fail (vs just printing a warning). Failure occurred for them consistently on macOS, but I wasn't able to replicate on any platform. I was only able to get mina-sshd to print a warning with a similar warning message.

This isn't an issue on the roborio because the roborio doesn't require authentication at all, so mina never tries to authenticate with a public key.

If the workaround below resolves it, this is an easy fix in deployutils.

### Potential temporary workaround (untested, will update)

In build.gradle, add the following:
Above `plugins` block:
```groovy
import edu.wpi.first.deployutils.deploy.target.location.SshDeployLocation;
import org.apache.sshd.common.keyprovider.KeyIdentityProvider
import edu.wpi.first.deployutils.deploy.sessions.AcceptAllLoggedServerKeyVerifier;
```
Inside the `deploy` block:
```groovy
deploy {
    targets {
        systemcore(getTargetTypeClass('SystemCore')) {
            // Team number is loaded either from the .wpilib/wpilib_preferences.json
            // or from command line. If not found an exception will be thrown.
            // You can use getTeamOrDefault(team) instead of getTeamNumber if you
            // want to store a team number in this file.
            team = project.frc.getTeamNumber()
            debug = project.frc.getDebugOrDefault(false)
			
			/* NEW LINES BELOW */
            locations.each { location ->
                if(location instanceof SshDeployLocation) {
                    location.setClientSessionConfiguration { session ->
                        session.keyIdentityProvider = KeyIdentityProvider.EMPTY_KEYS_PROVIDER
                        session.serverKeyVerifier = new AcceptAllLoggedServerKeyVerifier(logger)
                        if (location.password != null && !location.password.isBlank()) {
                            session.addPasswordIdentity(location.password);
                        }
                    }
                }
            }
			/* NEW LINES ABOVE */
// rest of file
```

AustinSchuh, 2025-08-07:
FYI, rsa keys appear to be deprecated.  I've seen super weird failures when using them.  Distros are slowly phasing them out as things propagate from development channels out to stable releases.  https://www.openssh.com/txt/release-8.2

AustinSchuh, 2025-08-07:
Ah, great.  At least one of my keys used the SHA1 signature.  I don't know about yours.