wpilibsuite/SystemcoreTesting #162: C++:  CTRE chassis commands don't seem to be sending signals to the motors

witcpalek, 2025-08-07:
Ported our CTRE generated chassis C++ code to system core, but it doesn't drive motors.

Specifically, ctre::phoenix6::swerve::requests::FieldCentric doesn't seem to be sending a signal to the motors.   We can see the command that executes this running, but the hoot logs are showing the value getting to the motors.   

We've run the motors with a simple TalonFX open loop to verify firmware and wiring is correct.

Is this a known issue with the CTR Phoenix X swerve generated code?

Next steps we are going to try:
- attempt to regenerate the swerve code (did something change and this class or some of the methods get deprecated; not seeing it in the doco, but maybe we are missing it).
- reverting to our prior custom swerve chassis code that isn't generated using the CTR swerve generator

bhall-ctre, 2025-08-07:
There shouldn't be any changes necessary from a 2025 generated swerve project other than swapping out the vendordep.

- Does the robot move in simulation?
- Is kCANBusName in TunerConstants correct?
    - Note that the onboard buses should be "can_s0" through "can_s4"
- What do the motors report for their RobotEnable and DeviceEnable status signals? (Can also be seen in Tuner X Self Test)
- If using CANivore, does the CANivore show up in Tuner X?

witcpalek, 2025-08-07:
@bhall-ctre 

Robot moves in simulation.  

The CAN bus name is set:
    static constexpr std::string_view kCANBusName = "can_s0";

I will need to check on the signals as the log is on someone else's computer. @Tan560 or @reslip1 do you have answers to the question about the Enable signals?

We don't have a CANivore in the system core set up (we had one on the RoboRio, but since the systemcore has CanFD we wired the motors directly to the can_s0 bus.

witcpalek, 2025-08-07:
@daltzctr and @bhall-ctre thank you for the quick responses.  We will check it out when we are back in our facility next week.

fangeugene, 2025-08-07:
> since the systemcore has CanFD we wired the motors directly to the can_s0 bus.

As I understand it, the systemcore CAN FD is not (currently) compatible with the CTRE flavor of it, meaning certain features (e.g. dynamic motion magic and time sync) don't actually work. Can someone from CTRE clarify if this is correct?

Tan560, 2025-08-08:
Here is the linker erorr, seems to be within the PathPlanner library. I was able to remove pathplanner and it was able to build and sim. I will try this on Monday next week and report back if this works on our chassisbot. 

> Task :linkFrcUserProgramLinuxsystemcoreReleaseExecutable FAILED
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\3fc7451164eaa97f87cf83e4a32852f9\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestatic\linux\systemcore\static\libPathplannerLib.a(FollowPathCommand.o): in function `pathplanner::FollowPathCommand::FollowPathCommand(pathplanner::FollowPathCommand&&)':
FollowPathCommand.cpp:(.text._ZN11pathplanner17FollowPathCommandC2EOS0_[_ZN11pathplanner17FollowPathCommandC5EOS0_]+0x58): undefined reference to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)'
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\3fc7451164eaa97f87cf83e4a32852f9\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestatic\linux\systemcore\static\libPathplannerLib.a(CommandUtil.o): in function `frc2::CommandHelper<frc2::Command, frc2::FunctionalCommand>::ToPtr() &&':
CommandUtil.cpp:(.text._ZNO4frc213CommandHelperINS_7CommandENS_17FunctionalCommandEE5ToPtrEv[_ZNO4frc213CommandHelperINS_7CommandENS_17FunctionalCommandEE5ToPtrEv]+0x5c): undefined reference to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)'
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\3fc7451164eaa97f87cf83e4a32852f9\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestatic\linux\systemcore\static\libPathplannerLib.a(AutoBuilder.o): in function `pathplanner::AutoBuilder::configure(std::function<frc::Pose2d ()>, std::function<void (frc::Pose2d const&)>, std::function<frc::ChassisSpeeds ()>, std::function<void (frc::ChassisSpeeds const&, pathplanner::DriveFeedforwards const&)>, std::shared_ptr<pathplanner::PathFollowingController>, pathplanner::RobotConfig, std::function<bool ()>, frc2::Subsystem*)::{lambda(std::shared_ptr<pathplanner::PathPlannerPath>, pathplanner::PathConstraints)#3}::operator()(std::shared_ptr<pathplanner::PathPlannerPath>, pathplanner::PathConstraints) const':
AutoBuilder.cpp:(.text+0x42bc): undefined reference to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)'
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\3fc7451164eaa97f87cf83e4a32852f9\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestatic\linux\systemcore\static\libPathplannerLib.a(AutoBuilder.o): in function `pathplanner::AutoBuilder::regenerateSendableReferences()':
AutoBuilder.cpp:(.text+0x490c): undefined reference to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)'
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\3fc7451164eaa97f87cf83e4a32852f9\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestatic\linux\systemcore\static\libPathplannerLib.a(AutoBuilder.o): in function `frc2::CommandHelper<frc2::Command, pathplanner::PathPlannerAuto>::ToPtr() &&':
AutoBuilder.cpp:(.text._ZNO4frc213CommandHelperINS_7CommandEN11pathplanner15PathPlannerAutoEE5ToPtrEv[_ZNO4frc213CommandHelperINS_7CommandEN11pathplanner15PathPlannerAutoEE5ToPtrEv]+0x5c): undefined reference to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)'
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\3fc7451164eaa97f87cf83e4a32852f9\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestatic\linux\systemcore\static\libPathplannerLib.a(AutoBuilder.o):AutoBuilder.cpp:(.text._ZNO4frc213CommandHelperINS_7CommandENS_15DeferredCommandEE5ToPtrEv[_ZNO4frc213CommandHelperINS_7CommandENS_15DeferredCommandEE5ToPtrEv]+0x5c): more undefined references to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)' follow
collect2.exe: error: ld returned 1 exit status

> Task :linkFrcUserProgramLinuxsystemcoreDebugExecutable FAILED
c:/users/public/wpilib/2027_alpha1/systemcore/bin/../aarch64-linux-gnu/sysroot/usr/lib/gcc/aarch64-linux-gnu/12/../../../../../../../aarch64-linux-gnu/bin/ld.exe: C:\Users\saxap\.gradle\caches\8.11\transforms\bc494e969e5ecba007f5234e00d9c704\transformed\PathplannerLib-cpp-2027.0.0-alpha-1-linuxsystemcorestaticdebug\linux\systemcore\static\libPathplannerLib.a(AutoBuilder.o): in function `wpi::SmallPtrSetImpl<frc2::Subsystem*>::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)':
C:\Users\runneradmin\.gradle\caches\8.11\transforms\a52ff625a9715490a52137a37cc6b429\transformed\wpiutil-cpp-2027.0.0-alpha-1-headers/wpi/SmallPtrSet.h:330: undefined reference to `wpi::SmallPtrSetImplBase::SmallPtrSetImplBase(void const**, unsigned int, wpi::SmallPtrSetImplBase&&)'
collect2.exe: error: ld returned 1 exit status

Tan560, 2025-08-08:
> That is correct, the SystemCore CAN buses do not support FD for this first round of alpha testing.

Wait sorry I missed these replies initially. Does this mean to run the chassis code we need to run with a canivore?

bhall-ctre, 2025-08-08:
Our swerve API does not require a CANivore, it just allows for 250 Hz control and odometry (CAN 2.0 only gets 100 Hz).

spacey-sooty, 2025-08-08:
> > If you can generate a new tuner swerve project, or grab our example and pull your tuner constants in, that would be useful to isolating it away from your code.
> > Last I tested (shortly before wpilib alpha 2 release), both Java and C++ swerve stacks work as expected on our office bot.
> 
> We tried pulling the C++ example (https://github.com/CrossTheRoadElec/Phoenix6-Examples/tree/main/cpp/SwerveWithPathPlanner) and we are getting link errors with patholanner. Is there another example that I’m not seeing?

Did you switch out the pathplanner vendordep for a 2027 one?

witcpalek, 2025-08-08:
Yes we are using the 2027 vendor dep

spacey-sooty, 2025-08-08:
you need to add systemcore to this list https://github.com/mjansen4857/pathplanner/blob/2027/examples/cpp/vendordeps/PathplannerLib.json#L33

witcpalek, 2025-08-08:
@spacey-sooty 

Thanks for the fix

spacey-sooty, 2025-08-08:
As mjansen mentioned in that PR it should be linuxsystemcore not systemcore

Tan560, 2025-08-11:
@bhall-ctre  I was able to confirm the CTRE example works on our System Core. Thank you for the help, I will continue to debug our code

Tan560, 2025-08-11:
Thanks everyone for responding, the issue with our code was a bug already found here (https://github.com/wpilibsuite/SystemCoreTesting/issues/37) The GetTeamNumber() Method always returned 0 and so we never created our chassis for our SwerveBot (Team number 9998). This issue can be closed out, thanks again!

closed 2025-08-12 by bhall-ctre