wpilibsuite/SystemcoreTesting #164: REV Magnetic Limit Switch not being read on SystemCore Smart I/O ports

taran-duba, 2025-08-16:
Hi all,

We are running a Rev Magnetic limit switch on our robot with SystemCore and are having an issue with the boolean reading not changing. We’ve tested on multiple ports, verified the mag switch is detecting the magnet, verified the signal wire changes voltage (3.3V) with the magnet, verified it works on the rio with a test board. There’s no change in boolean signal when connected to SystemCore. 

We’ve read here that the internal resistor is a 4.7kOhm pull down resistor in syscore vs a 40kOhm pull up resistor on the rio. when running power to syscore the multimeter reads 3.3 volts. we think this should just mean the value is inverted? but it may be causing this lack of value changing too.

Has anyone experienced this before?

samperlmutter, 2025-08-16:
Hi I am the mentor from this student's team (3506). Just wanted to add that the [spec sheet for the mag switch](https://docs.revrobotics.com/rev-crossover-products/sensors/magnetic-limit-switch/specs) says 3.3V should be fine, which is why we think the resistor might be the culprit, but I'm not an EE so not certain.

samperlmutter, 2025-08-16:
Gotcha, that confirms what I was thinking. We are going to try a beam break or physical limit switch now. Thanks