wpilibsuite/SystemcoreTesting #166: CAN bus died on code deploy/code initialization

Xanawatt, 2025-08-21:
Upon deploying code, and subsequent code initialization, `can_s1` is unresponsive. ReduxLib appears to crash and take down `can_s1` with it. Upon rebooting, `can_s1` was recovered and operating like normal. We did notice that the Canandcolor we were using was powered off after the reboot fixed the issue, but it did not crash upon reboot.

We are using ReduxLib 2027.0.0-alpha-2 with a Canandcolor and Alpha 5 Release 166. Logs below:

(dmesg and canbusload taken prior to reboot)
```
$ dmesg
[output snipped]
[ 1079.119364] CAN Heartbeat: Enabled state changed to 1
[ 1107.439821] CAN Heartbeat: Enabled state changed to 0
[ 1211.580710] CAN Heartbeat: Enabled state changed to 1
[ 1247.159973] CAN Heartbeat: Enabled state changed to 0
[ 1287.541464] CAN Heartbeat: Enabled state changed to 1
[ 1289.580191] CAN Heartbeat: Enabled state changed to 0
[ 1821.073740] kauditd_printk_skb: 116 callbacks suppressed
[ 1821.073747] audit: type=1701 audit(1755809733.391:48): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1002 comm=564D20546872656164 exe="/usr/lib/jvm/bin/java" sig=6 res=1
[ 1823.982543] audit: type=1334 audit(1755809736.303:49): prog-id=11 op=LOAD
[ 1823.982610] audit: type=1334 audit(1755809736.303:50): prog-id=8 op=UNLOAD
[ 1823.982773] audit: type=1334 audit(1755809736.303:51): prog-id=12 op=LOAD
[ 1823.982885] audit: type=1334 audit(1755809736.303:52): prog-id=13 op=LOAD
[ 1823.983003] audit: type=1334 audit(1755809736.303:53): prog-id=9 op=UNLOAD
[ 1823.983027] audit: type=1334 audit(1755809736.303:54): prog-id=10 op=UNLOAD
[ 1823.987278] audit: type=1334 audit(1755809736.307:55): prog-id=14 op=LOAD
[ 1823.987524] audit: type=1334 audit(1755809736.307:56): prog-id=15 op=LOAD
[ 1823.987679] audit: type=1334 audit(1755809736.307:57): prog-id=6 op=UNLOAD
[ 1862.442087] CAN Heartbeat: Enabled state changed to 1
[ 1867.353550] mcp251xfd spi2.1 can_s1: IRQ handler mcp251xfd_handle_tefif() returned -22.
[ 1867.353587] mcp251xfd spi2.1 can_s1: IRQ handler returned -22 (intf=0x3f1aa014).
[ 1883.361208] CAN Heartbeat: Enabled state changed to 0
[ 1889.761039] CAN Heartbeat: Enabled state changed to 1
[ 1900.181583] CAN Heartbeat: Enabled state changed to 0
```

```
$ canbusload can_s1@1000000 -ertbc
canbusload 2025-08-21 17:04:09 (exact bitstuffing)
 can_s1@1000000     0       0      0      0   0% |....................|
```

```
Aug 21 16:55:32 robot sshd[65251]: Accepted keyboard-interactive/pam for systemcore from 10.4.61.209 port 56238 ssh2
Aug 21 16:55:32 robot sshd-session[65251]: pam_unix(sshd:session): session opened for user systemcore(uid=105) by (uid=0)
Aug 21 16:55:32 robot bash[1002]: The timestamp reported by CANivore is at least 10ms older than the timestamp reported by the system, indicating it's fallen out of sync. This does not
Aug 21 16:55:32 robot bash[1002]:         ctre::phoenix6::BaseStatusSignal::RefreshAll
Aug 21 16:55:32 robot MrcCommDaemon[313]: Sending system info 1840267264 3838238720 43.333332
Aug 21 16:55:33 robot sudo[65280]: systemcore : PWD=/home/systemcore ; USER=root ; COMMAND=/bin/systemctl stop robot
Aug 21 16:55:33 robot sudo[65280]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=105)
Aug 21 16:55:33 robot systemd[1]: Stopping robot...
Aug 21 16:55:33 robot bash[1002]: [ReduxLib] CanandEventLoop exit.
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot bash[1002]: # A fatal error has been detected by the Java Runtime Environment:
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot bash[1002]: #  SIGSEGV (0xb) at pc=0x0000007f8c8a17c4, pid=1002, tid=1017
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot bash[1002]: # JRE version: OpenJDK Runtime Environment (21.0+21) (build 21+21)
Aug 21 16:55:33 robot bash[1002]: # Java VM: OpenJDK 64-Bit Server VM (21+21, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, linux-aarch64)
Aug 21 16:55:33 robot bash[1002]: # Problematic frame:
Aug 21 16:55:33 robot bash[1002]: # C  [libc.so.6+0x837c4]  pthread_mutex_lock+0x4
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot bash[1002]: # Core dump will be written. Default location: Core dumps may be processed with "/bin/false" (or dumping to /home/systemcore/core.1002)
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot bash[1002]: # An error report file with more information is saved as:
Aug 21 16:55:33 robot bash[1002]: # /home/systemcore/hs_err_pid1002.log
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot bash[1002]: # If you would like to submit a bug report, please visit:
Aug 21 16:55:33 robot bash[1002]: #   https://bugreport.java.com/bugreport/crash.jsp
Aug 21 16:55:33 robot bash[1002]: #
Aug 21 16:55:33 robot kernel: kauditd_printk_skb: 116 callbacks suppressed
Aug 21 16:55:33 robot kernel: audit: type=1701 audit(1755809733.391:48): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1002 comm=564D20546872656164 exe="/usr/lib/jvm/bin/java" sig=6 r
Aug 21 16:55:33 robot audit[1002]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1002 comm=564D20546872656164 exe="/usr/lib/jvm/bin/java" sig=6 res=1
[... 137 lines cut ...]
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.RobotContainer.<init>(RobotContainer.java:26)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.Robot.<init>(Robot.java:18)
Aug 21 16:55:39 robot bash[65514]:         at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:268)
Aug 21 16:55:39 robot bash[65514]:         at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:388)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.Main.main(Main.java:9)
Aug 21 16:55:39 robot bash[65514]: Error at com.reduxrobotics.canand.CanandSettingsManager.setSettings(CanandSettingsManager.java:261): 5 settings could not be applied to Canandcolor[d
Aug 21 16:55:39 robot bash[65514]:         at com.reduxrobotics.canand.CanandSettingsManager.setSettings(CanandSettingsManager.java:261)
Aug 21 16:55:39 robot bash[65514]:         at com.reduxrobotics.sensors.canandcolor.Canandcolor.setSettings(Canandcolor.java:483)
Aug 21 16:55:39 robot bash[65514]:         at com.reduxrobotics.sensors.canandcolor.Canandcolor.setSettings(Canandcolor.java:494)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.subsystems.intake.Intake.<init>(Intake.java:69)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.RobotStates.<init>(RobotStates.java:61)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.RobotContainer.<init>(RobotContainer.java:26)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.Robot.<init>(Robot.java:18)
Aug 21 16:55:39 robot bash[65514]:         at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:268)
Aug 21 16:55:39 robot bash[65514]:         at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:388)
Aug 21 16:55:39 robot bash[65514]:         at frc.robot.Main.main(Main.java:9)
Aug 21 16:55:39 robot bash[65514]: Could not transmit CAN Frame.
Aug 21 16:55:39 robot bash[65514]:         cancoder 62 ("can_s1") Apply Config
Aug 21 16:55:39 robot MrcCommDaemon[313]: Sending system info 1840041984 3929251840 41.935486
Aug 21 16:55:40 robot bash[65514]: [phoenix] Library initialization is complete.
Aug 21 16:55:40 robot bash[65514]:         
Aug 21 16:55:40 robot bash[65514]: ********** Robot program startup complete **********
```

CC @Quincy-T

rzblue, 2025-08-21:
The error messages from reduxlib appear to be a symptom of the interface not working rather than the cause.

`mcp251xfd_handle_tefif() returned -22` is suspicious, wonder if this is related at all to #163

rzblue, 2025-09-28:
The message from the mcp251xfd driver matches the one in this issue: https://github.com/raspberrypi/linux/issues/6644