wpilibsuite/SystemcoreTesting #168: Kill robot code step of Deploy takes 90 seconds

chauser, 2025-08-29:
We have observed that deploying code from VSCode takes much longer than expected.  We have alpha 5 OS and alpha 2 WPILib. We had noticed that clicking restart robot code from the driverstation allows code deploy to proceed immediately.

We finally took some time to track it down tonight. The step that takes a long time is killing the running robot code. This is done in the deploy script by using the command `systemctl stop robot`. Running that command from an ssh shell has the same behavior -- it reliably takes about 90 seconds to run if there is active robot code. If there is no running robot code the `stop` command returns instantly.

`systemctl stop` is documented to send SIGTERM and, if the service hasn't stopped after 90 seconds, then send SIGKILL. This is most likely what is going on.

Maybe a known problem but I haven't found any mention of it in the places I've looked.

ThadHouse, 2025-08-29:
Agree we should set the timeout lower, but this is a tell tale sign something in your project is locking on shutdown.

chauser, 2025-09-25:
As a user and not a developer, my opinion is that handling the code restart synchronously with DS packet processing is probably OK as long as the KILL signal is sent reasonably promptly after the code restart button is clicked. On the roboRIO it looks like the delay from sending TERM to sending KILL is 1 second. It is not clear to me what advantage would be gained by doing the restart actions outside the packet handling loop: yes you could do other things while waiting for the restart to finish but whatever those things were, they would be asynchronous with the restart and those asynchronous behaviors and possible misbehaviors would all need to be considered/specified/tested/debugged.