wpilibsuite/SystemcoreTesting #171: External Python client cannot complete the NetworkTables handshake on SystemCore

kalsing, 2025-09-21:
We are trying to use a Python script that worked in 2025 to send keyboard inputs through NetworkTables. With SystemCore we are not sure how external clients should connect, or if this is a limitation of SystemCore or our setup.

Current behavior: the client starts but does not complete the NT handshake and no entries appear on the robot side. Ping to the SystemCore IP succeeds.

The python script we are running is the following:
```
import networktables as networkTablesCore
import keyboard
import time
from threading import Timer

def main():
    def release_key(key_name: str) -> None:
        print(f"Sending: {key_name} -> {False}")
        table.putBoolean(key_name, False)

    def timed_keypress(key_name: str, press_time: float) -> None:
        print(f"Sending: {key_name} -> {True}")
        table.putBoolean(key_name, True)
        timer = Timer(press_time, release_key, [key_name])
        timer.start()

    def on_action(event: keyboard.KeyboardEvent):
        if (event.name == "/"):
            return
        value = event.event_type == keyboard.KEY_DOWN
        if value:
            key_name = "numpad" + event.name if event.is_keypad else event.name.lower()
            timed_keypress(key_name, 0.100)

    networkTables = networkTablesCore.NetworkTablesInstance.getDefault()

    print("Setting up NetworkTables client")
    networkTables.startClient("KeyboardToNT")
    networkTables.setServer("10.11.56.2Firstly, we're not certain if this is an issue of the SystemCore.
We're encountering an issue where NetworkTables no longer connects when running our Python script from the SystemCore. This script used to work on our 2025 robot to send keyboard input via NetworkTables, and we're sure the IP is correct. We're not certain if the problem is because of newer changes in NetworkTables. Does anyone know how external clients are currently expected to connect to NetworkTables?

This is the python script we're running
[KeyboardToNetworkTables.py](https://github.com/user-attachments/files/22453018/KeyboardToNetworkTables.py)
")
    networkTables.startDSClient()

    print("Waiting for connection to NetworkTables server...")
    while not networkTables.isConnected():
        time.sleep(0.1)

    print("Connected!")
    table = networkTables.getTable("OperatorController")

    keyboard.hook(lambda e: on_action(e))
    keyboard.wait()

if __name__ == '__main__':
    main()
```

ThadHouse, 2025-09-21:
Just to double check, do you have the DS running?

And then after that, did you set a static IP for the robot? If you, can you try setting the IP away from .2 to whatever the radio assigned to it? The screen will show you what the IP is.

kalsing, 2025-09-21:
Yes, same machine for DS and the Python client, with the SystemCore at the 10.11.56.2 IP.

We set the systemcore to a static IP of 10.11.56.20, whitch matches what the SystemCore screen showed after switching SystemCore to DHCP. Still no connection with the Python client.

ThadHouse, 2025-09-21:
You should set the static IP to .2. That will match what the radio gives it when new radio firmware comes. 

Also, just to double check, what version of NetworkTables. Is it new enough to be NT4 based? NT3 support was removed from 2027 WPILib.

PeterJohnson, 2025-09-21:
PyNetworkTables is NT3.  The 2027 library is NT4 only.  You’ll need to switch your Python code to use pyntcore.

virtuald, 2025-09-21:
you need to upgrade to pyntcore. pynetworktables doesn't support NT4.