wpilibsuite/SystemcoreTesting #176: SystemCore Dangerous Communication Error

Raphdf201, 2025-09-28:
We had a big issue with the SystemCore communication. The issue is that sometimes, when the SystemCore loses wifi connection with the Driver Station, it stays enabled and keeps the last joystick input. We use the integrated access point of the SystemCore. Combined with the fact that the SystemCore often loses communication, it makes the issues really DANGEROUS. Please be cautious!!!

[ChiefDelphi post](https://www.chiefdelphi.com/t/systemcore-dangerous-communication-error/506661)
[Video](https://drive.google.com/file/d/1Y0MgkbtdVcWvYA65F1WJLUjf1r0AOIBb/view)

Motors : Kraken x60 with firmware 25.90.0.0
Vendordeps : Phoenix6 25.90.0-alpha-1
                      PathplannerLib 2027.0.0-alpha-2
                      WPILib-New-Commands 1.0.0
DS:  25.0
Systemcore version :  limelightosr-alpha-6
WPILib version : 2027_alpha1

WiFi settings
[image]

ThadHouse, 2025-09-28:
Can you reproduce it on alpha 5? Alpha 6 (build 167) was purely a testing release, and wasn't for teams to really use.

Arnaud-5G, 2025-09-28:
> Can you reproduce it on alpha 5? Alpha 6 (build 167) was purely a testing release, and wasn't for teams to really use.

Sure we will also test on another computer with newly installed 2027 software.

Raphdf201, 2025-09-28:
We tested on another computer with alpha6, and the same computer as before with alpha5 and weren't able to reproduce the issue

ThadHouse, 2025-09-28:
Interesting. And then if you go back to alpha 6 and the old laptop does it reproduce?

Raphdf201, 2025-09-28:
We will try that tomorrow

ThadHouse, 2025-10-02:
We found the root cause of this issue. However it's not exclusive to Alpha 6. It can happen on any Alpha OS build. Working on a fix, but just a note to be vigilant even on old builds. 

Note only disconnects are affected. If you properly disable you won't run into the issue.

Raphdf201, 2025-10-03:
> Interesting. And then if you go back to alpha 6 and the old laptop does it reproduce?

Yes